OpenInverterGateway/OpenInverterGateway #465: Can not compile

JohnDoeChillt, 2025-03-10:
Hello,
just installed VS Code and platformio (clean install).

(I only need a GrowattTLXH.cpp for the Shine wifiX with Modbus writing enabled)

When building the image there are very much issues.

1. lib_deps offline
solved with: https://github.com/plieven/ModbusMaster#16d5f1f1cb4e1c9a7943425557a09ac9e89d624c

2. Very very Much compiling issues, don´t know what to do. I can´t even copy&paste it here bcs its too much, so here a little example:

. c:\users\xxx\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_multimap.h:660:63: error: wrong number of template arguments (2, should be 5)
  660 |    using _Merge_helper = _Rb_tree_merge_helper<multimap, _Cmp2>;
      |                                                               ^
c:\users\xxx\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_multimap.h: In member function 'std::multimap<_Key, _Tp, _Compare, _Alloc>::iterator std::multimap<_Key, _Tp, _Compare, _Alloc>::insert(int&&)':
c:\users\xxx\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_multimap.h:646:16: error: '_M_t' was not declared in this scope
  646 |       { return _M_t._M_reinsert_node_equal(std::move(__nh)); }
      |                ^~~~
c:\users\xxx\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_multimap.h: In member function 'std::multimap<_Key, _Tp, _Compare, _Alloc>::iterator std::multimap<_Key, _Tp, _Compare, _Alloc>::insert(std::multimap<_Key, _Tp, _Compare, _Alloc>::const_iterator, 
std::multimap<_Key, _Tp, _Compare, _Alloc>::value_type&&)':
c:\users\xxx\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_multimap.h:590:16: error: '_M_t' was not declared in this scope

ChrisHVL, 2025-03-11:
Similar to my experience.
When it was based on Arduino IDE it run better.

rgds de DD6CN

crasu, 2025-03-11:
You should not fiddle around with libs_deps.

Just triggerd a ci build [here](https://github.com/OpenInverterGateway/OpenInverterGateway/actions/runs/13782666438/job/38543625867). Works without issues. The ci build is using platformio-6.1.17 on linux. 

Could be an OS or version issue.

closed 2025-03-11 by crasu