OpenInverterGateway/OpenInverterGateway #466: Set settings

Dane8991, 2025-03-17:
Hey there is it possible to set the inverter settings over Mqtt for example? I want to set the ACCharge setting true/false via nodered.

Best regards,
Dane

perosb, 2025-03-18:
This question probably belongs to "Discussions" rather than an issue.

And, the answer is already in the documentation: https://github.com/OpenInverterGateway/OpenInverterGateway/blob/master/Doc/MQTT.md#version-for-protocol-124

Dane8991, 2025-03-21:
Thanks for your answers. I know the MQTT.md but there is no example for set battery. The set/get date works fine but I don´t know what the firmware expects as payload for battery/get and so on

closed 2025-03-21 by Dane8991

perosb, 2025-03-21:
@Dane8991 

These are my home-assistant mqtt `number` entites. You can see the command_topic settings a number value and the `switch` changing AC charging.

```yaml
mqtt:
  switch:
    - name: "growatt_bdcchargeacenabled"
      unique_id: growatt_bdcchargeacenabled
      payload_on: "1"
      payload_off: "0"
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCChargeACEnabled'] }}"
      command_topic: "energy/solar/command/bdc/set/acchargeenabled"
      command_template: '{"value": {{ value }} }'
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"
   

  number:
    - name: "growatt_bdcchargestopsoc"
      unique_id: growatt_bdcchargestopsoc
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCChargeStopSOC'] }}"
      icon: mdi:battery-plus-variant
      command_topic: "energy/solar/command/bdc/set/chargestopsoc"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

    - name: "growatt_bdcdischargestopsoc"
      unique_id: growatt_bdcdischargestopsoc
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCDischargeStopSOC'] }}"
      icon: mdi:battery-minus-variant
      command_topic: "energy/solar/command/bdc/set/dischargestopsoc"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 10
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

    - name: "growatt_bdcdischargepowerrate"
      unique_id: growatt_bdcdischargepowerrate
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCDischargePowerRate'] }}"
      command_topic: "energy/solar/command/bdc/set/dischargepowerrate"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

    - name: "growatt_bdcchargepowerrate"
      unique_id: growatt_bdcchargepowerrate
      state_topic: "energy/solar"
      value_template: "{{ value_json['BDCChargePowerRate'] }}"
      command_topic: "energy/solar/command/bdc/set/chargepowerrate"
      command_template: '{"value": {{ value }} }'
      unit_of_measurement: '%'
      min: 0
      max: 100
      step: 5
      availability_mode: latest
      availability_topic: "energy/solar"
      availability_template: "{{- '1' if value_json.InverterStatus|int(0) >= 0 else '0' -}}"
      payload_available: "1"
      device:
        identifiers:
          - "growatt"

```