OpenInverterGateway/OpenInverterGateway #471: Add support for export limit setting

lolimpol, 2025-03-23:
*Feature request*
An export limit setting would be a good addition for TLH-X inverters. It is currently the only setting one might use semi-regularly which is available on the app, and not on openinvertergateway. As mentioned [here](https://github.com/esphome/feature-requests/issues/2159) it is already supported in RTU v1.24:

[image]

gonzo7734, 2025-04-15:
I would love this feature, too.

But I think that the holding registers 122 and 123 are only for those, who use an external RS485 / RS232 energy meter or a CT (clamp terminal). Via touch screen (growatt mod series) you can already limit your export rate to x% of your max output power.

But if you have for example victron venus os (or any other system) with it's own energy meter and you have already integrated your growatt inveter with OpenInveterGateway AND don't want to buy additional hardware, then you should limit your growatt export power with holding registers 03 (and / or 04):

[image]

PS: you can already limit this via http://<ip>/postCommunicationModbus 
First you need to compile this repository on your own with:
`#define ENABLE_MODBUS_COMMUNICATION 1`
and second you have to build a logic in your home automation system that sends POST data to the link above.

Manual example:

[image]
gives me this result:

[image]

[image]

crasu, 2025-05-07:
You can set the active rate via mqtt see #474.

closed 2025-05-07 by crasu