OpenInverterGateway/OpenInverterGateway #474: Set/Get Active Rate via MQTT

stephanDE, 2025-04-07:
Hello, 
everything works like a charme, but i want to set/get the export rate of my Growatt TLXH 2500 via MQTT,
and home assistant:
```
hass.services.call("mqtt", "publish", {
    "topic": "energy/solar/command/activerate/get"
})
```

i get as response in my mqtt explorer:

`result = {"command":"activerate/get","success":false,"message":"Failed to parse JSON request: EmptyInput"}
`

and to set the active rate i do
```
hass.services.call("mqtt", "publish", {
    "topic": "energy/solar/command/activerate/set",
    "payload":'{"value": 20}',
})
```
getting as response:
```
`result = {"command":"activerate/set","success":false,"message":"Unknown command: activerate/set"}
`

it would be great to get some hints how to do this,

HENN0, 2025-04-21:
@stephanDE  Any update?

jeroen-github, 2025-04-24:
- service: mqtt.publish
            data:
              topic: energy/solar/command/power/set/activerate
              payload: "{{ {'value': 50} }}"

stephanDE, 2025-04-28:
Thank you, that work, the only thing is: decreasing power happens immediately. But increasing the power takes some time, increasing from 20% to 100% takes about 1 minute. Is this normal ?

HENN0, 2025-04-30:
Lowering the power is easy for the inverter, but increasing takes some time to find mppt
The same happens if you turn off PV input, it will take over 1 minute till it is at the same value as before turning off

My control to adapt to the needed power works nice, just the stupid smart meter disable its output every 1-2 days and my provider dosnt want to help

closed 2025-05-07 by crasu