OpenInverterGateway/OpenInverterGateway #477: RW Modbus - Not found: /postCommunicationModbus

pge, 2025-04-24:
Im using the ShinewifiX ith OpenInverterFirmware.
Unfortunately trying to use the RW Modbus feature in the webinterface leads to the error 

Not found: /postCommunicationModbus

Any idea on that?

mdk2412, 2025-04-24:
This is not an issue!

It is simply not included (configured) in the release versions. You have to define it in the config.h and compile yourself a version having it.

PP1723, 2025-04-26:
I have the same problem. I don't know how to do this exactly. Is there somewhere a tutorial how to do this?
Sorry I'm totally new to this

pge, 2025-04-27:
It indeed is not an issue. Thanks for clearing that uo @mdk2412!
I compiled a version with mqtt and modbus_rw enabled. 
@PP1723 See here:  https://cloud.openroot.io/s/8iK5c2LoQxQDFia

closed 2025-04-27 by pge

muhkuh666, 2025-05-06:
> This is not an issue!
> 
> It is simply not included (configured) in the release versions. You have to define it in the config.h and compile yourself a version having it.

How to do so? Please explain